pablomarquez76/PS4_Controller_Host #1: PS4 Gamepad not connecting/pairing.

Gustavomurta, 2023-12-23:
It wasn't clear to me, but I changed the MAC address in the program.
PS4.begin("00:1f:e2:ac:e9:27"); //  PS4 gamepad mac address

I have tried pair it with a ESP32 (hold PS and share at the same time until the light blinks twice in quick succession rapidly). 
But it´s not working. No paired, no connected. 

I obtained the MAC address of the gamepad by connecting and pairing in Windows. (capturing the data with Wireshark).
[Source: HonHaiPrecis_ac:e9:27 (00:1f:e2:ac:e9:27)]

Can you clarify how to connect, please?

pablomarquez76, 2023-12-24:
Hi Gustavo,

In order to connect your controller to the ESP32, you need to match the MAC address of you ESP32 to the MAC address stored in the PS4 controller. Another option is to change the MAC address stored in your controller to match the MAC address of your ESP. The MAC address stored in the controller (which is different to the MAC address of the controller) tells the controller which device it must connect. 

The function PS4.begin("new MAC address of your ESP32); basically changes the MAC address of your ESP32. So, for example, if your controller is linked to your phone, you need to change the MAC address of your ESP to the MAC address of your phone. Or, you can use SixaxisPairTool program (or similar) to change the MAC address stored in your controller to match the MAC address of your ESP (in this case, the code to begin the controller must be PS4.begin(); ).

This diagram may help you to understand how this works:

Gamepad MAC     (Gamepad stored MAC)       <>             ESP32 MAC
11:11:11:11:11:11  (22:22:22:22:22:22)               <>             22:22:22:22:22:22

I hope this clarifies your issue.

Gustavomurta, 2023-12-24:
Hi Pablo, 
Thanks for the quick response.
I do not intend to change the MAC Address of the my PS4 gamepad.
"In order to connect your controller to the ESP32, you need to match the MAC address of you ESP32 to the MAC address stored in the PS4 controller.The function PS4.begin("new MAC address of your ESP32); basically changes the MAC address of your ESP32."

I included my Gamepad's MAC address in this line.
**PS4.begin("00:1f:e2:ac:e9:27");** // PS4 gamepad mac address

I didn't understand what I did wrong.

pablomarquez76, 2023-12-24:
Sometimes the problem is that the memory assigned for MAC addresses of paired devices in the ESP32 is full. In that case, you should use the function removePairedDevices() to free the memory before uploading you code. So you may take that into account.