pablomarquez76/PS4_Controller_Host #3: Trouble making a datafrog PS4 controller connect to ESP

cvjensen, 2024-02-16:
Hi @pablomarquez76  , 

Am I missing something or does this PS4 controller replica just not work?

I have uploaded the following to my ESP32 (just a normal edition)

```
#include <PS4Controller.h>
#include "esp_bt_main.h"
#include "esp_bt_device.h"
#include "esp_gap_bt_api.h"
#include "esp_err.h"

unsigned long lastTimeStamp = 0;
#define EVENTS 0
#define BUTTONS 0
#define JOYSTICKS 1
#define SENSORS 0

void removePairedDevices() {
  uint8_t pairedDeviceBtAddr[20][6];
  int count = esp_bt_gap_get_bond_device_num();
  esp_bt_gap_get_bond_device_list(&count, pairedDeviceBtAddr);
  for (int i = 0; i < count; i++) {
    esp_bt_gap_remove_bond_device(pairedDeviceBtAddr[i]);
  }
}

void printDeviceAddress() {
  const uint8_t* point = esp_bt_dev_get_address();
  for (int i = 0; i < 6; i++) {
    char str[3];
    sprintf(str, "%02x", (int)point[i]);
    Serial.print(str);
    if (i < 5) {
      Serial.print(":");
    }
  }
}

void onConnect() {
  Serial.println("Connected!");
}

void notify() {
#if EVENTS
  boolean sqd = PS4.event.button_down.square,
          squ = PS4.event.button_up.square,
          trd = PS4.event.button_down.triangle,
          tru = PS4.event.button_up.triangle;
  if (sqd)
    Serial.println("SQUARE down");
  else if (squ)
    Serial.println("SQUARE up");
  else if (trd)
    Serial.println("TRIANGLE down");
  else if (tru)
    Serial.println("TRIANGLE up");
#endif

#if BUTTONS
  boolean sq = PS4.Square(),
          tr = PS4.Triangle();
  if (sq)
    Serial.print(" SQUARE pressed");
  if (tr)
    Serial.print(" TRIANGLE pressed");
  if (sq | tr)
    Serial.println();
#endif

  //Only needed to print the message properly on serial monitor. Else we dont need it.
  if (millis() - lastTimeStamp > 50) {
#if JOYSTICKS
    Serial.printf("lx:%4d,ly:%4d,rx:%4d,ry:%4d\n",
                  PS4.LStickX(),
                  PS4.LStickY(),
                  PS4.RStickX(),
                  PS4.RStickY());
#endif
#if SENSORS
    Serial.printf("gx:%5d,gy:%5d,gz:%5d,ax:%5d,ay:%5d,az:%5d\n",
                  PS4.GyrX(),
                  PS4.GyrY(),
                  PS4.GyrZ(),
                  PS4.AccX(),
                  PS4.AccY(),
                  PS4.AccZ());
#endif
    lastTimeStamp = millis();
  }
}

void onDisConnect() {
  Serial.println("Disconnected!");
}

void setup() {
  Serial.begin(115200);
  PS4.attach(notify);
  PS4.attachOnConnect(onConnect);
  PS4.attachOnDisconnect(onDisConnect);
  PS4.begin();
  removePairedDevices(); // This helps to solve connection issues
  Serial.print("This device MAC is: ");
  printDeviceAddress();
  Serial.println("");
}

void loop() {
  delay(100);
}

```

Then it says:

`This device MAC is: c8:f0:9e:2e:1b:66`

which I have copied and put into sixaxispairtool and changed the stored address.

Then I've tried to hold down various buttons/combinations to make the controller connect but without luck. 

Am I missing something or do I just have a bad controller?

pablomarquez76, 2024-02-16:
Hi @cvjensen,

If you change the PS4 controller to that of your ESP32 using sixaxispairtool, which will appear printed in the Serial Monitor after you upload that code, e. g.:

This device MAC is: c8:f0:9e:2e:1b:66

Then, you just press the PS button of your controller; it will blink for about 2 or 3 seconds. After that, when they connect, the light should stay on as long as the devices are connected.

Some alternative controllers work with PS4 but don´t work with this library.

I hope this help you.

cvjensen, 2024-02-17:
Thanks for the quick response.

Sixaxispairtool and "this device mac" are the same.

When i press the PS button it start blinking but I never get any steady light. It just goes out.

cvjensen, 2024-02-25:
are there any way to get a verbose output from your code ?

pablomarquez76, 2024-02-25:
You could set the "Core Debug Level" in your compiler to "Verbose".
I think you can also add at the beginnig of your code:
#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
#include "esp_log.h"