pablomarquez76/PS4_Controller_Host #4: Not working with esp32-s3

KyranHU, 2024-03-02:
Hi, I'm trying to use this library with an esp32-s3, but I get these errors when trying to build the code:

```
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\main.cpp.o
Linking .pio\build\esp32-s3-devkitc-1\firmware.elf
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o):(.literal.ps4_l2cap_disconnect_ind_cback+0x4): undefined reference to `L2CA_DisconnectRsp'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o):(.literal.ps4_l2cap_config_ind_cback+0x0): undefined reference to `L2CA_ConfigRsp'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o):(.literal.ps4_l2cap_connect_ind_cback+0x8): undefined reference to `L2CA_ErtmConnectRsp'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o):(.literal.ps4_l2cap_connect_ind_cback+0xc): undefined reference to `L2CA_ConfigReq'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o):(.literal.ps4_l2cap_init_service+0x14): undefined reference to `L2CA_Register'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o):(.literal.ps4_l2cap_send_hid+0x14): undefined reference to `L2CA_DataWrite'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o): in function `ps4_l2cap_disconnect_ind_cback':
C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_l2cap.c:280: undefined reference to `L2CA_DisconnectRsp'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o): in function `ps4_l2cap_config_ind_cback':
C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_l2cap.c:262: undefined reference to `L2CA_ConfigRsp'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o): in function `ps4_l2cap_connect_ind_cback':
C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_l2cap.c:193: undefined reference to `L2CA_ErtmConnectRsp'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_l2cap.c:196: undefined reference to `L2CA_ErtmConnectRsp'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_l2cap.c:199: undefined reference to `L2CA_ConfigReq'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o): in function `ps4_l2cap_init_service':
C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_l2cap.c:150: undefined reference to `L2CA_Register'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_l2cap.c.o): in function `ps4_l2cap_send_hid':
C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_l2cap.c:120: undefined reference to `L2CA_DataWrite'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_spp.c.o):(.literal.sppCallback+0x8): undefined reference to `esp_bt_gap_set_scan_mode'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_spp.c.o):(.literal.sppCallback+0xc): undefined reference to `esp_spp_start_srv' 
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_spp.c.o):(.literal.sppInit+0x14): undefined reference to `esp_spp_register_callback'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_spp.c.o):(.literal.sppInit+0x18): undefined reference to `esp_spp_init'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_spp.c.o): in function `sppCallback':
C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_spp.c:87: undefined reference to `esp_bt_gap_set_scan_mode'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_spp.c:92: undefined reference to `esp_spp_start_srv'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/lib775/libPS4Controller.a(ps4_spp.c.o): in function `sppInit':
C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_spp.c:56: undefined reference to `esp_spp_register_callback'
c:/users/kyran/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kyran\Documents\PlatformIO\Projects\Raven/.pio/libdeps/esp32-s3-devkitc-1/PS4Controller/src/ps4_spp.c:65: undefined reference to `esp_spp_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1
```

Can I get any help with this?
Thanks!

pablomarquez76, 2024-03-03:
Hi, the code is written only for ESP32 microcontrollers, but it doesn't support ESP32-S2, ESP32-C3 or ESP32S3 microcontrollers.

KyranHU, 2024-03-10:
That's unfortunate, thanks!

closed 2024-03-10 by KyranHU

masa8010, 2024-06-03:
ESP32-S3に対応したコードはありますか？