pablomarquez76/PS4_Controller_Host #6: Not working with esp32-s3

masa8010, 2024-06-03:
Is there a corresponding code for esp5-S3 in M3Atom S3?

pablomarquez76, 2024-06-03:
I don´t have any experience with that micro, if it has bluetooth classic the code may work.

masa8010, 2024-06-06:
Is it easy to develop?

pablomarquez76, 2024-06-06:
That a tricky question. If you have the knowledge and the board has the connectivity needed for connecting to a DS4 it may be easy. Otherwise, it may be impossible.

pablomarquez76, 2024-06-07:
> What kind of connectivity is needed to connect a board to a DS4?
DS4 uses bluetooth 2.1, I think some boards don´t have the support for this protocol which came out in 2007. They use BLE instead.

pablomarquez76, 2024-06-07:
> Or is there any other way to connect to the DS4?
There are some libraries that connect to DS4 through USB cable. This is not very useful if you want to control a robot wirelessly.

pablomarquez76, 2024-06-07:
If you want to know more about the DS4 protocol you may take a look at: https://www.psdevwiki.com/ps4/DS4-BT

masa8010, 2024-06-09:
I am not familiar with software and would like to ask for help.

pablomarquez76, 2024-06-10:
> I would like you to do the development, is it possible?

At this time, I'm working on many projects. Therefore, I don't have the time.

masa8010, 2024-06-11:
I understood. Wait patiently.

masa8010, 2024-06-11:
I understood. Ask a friend to help you.

masa8010, 2024-06-11:
I understood.