pablomarquez76/PS4_Controller_Host #10: use bluetooth and wifi and camera simultaneously?

drunkscientist, 2025-02-13:
so i just finished trying with my 4th esp-cam firmware, and all 4 work just fine up until PS4.begin, at which point either the camera output stops or wifi does. my first try was with an rtsp server, and i got way farther with it than i should have, since i wasnt verifing that the camera was still working as i wrote a bunch of control code. after getting all the controls working i realized the camera no longer had usable output. i tried with another firmware and found that the wifi either crashed or never properly activated, depending on if PS4.begin was in the setup or tied to another function. i dont actually know if this is an issue with the library or a limitation of the esp, but it is currently the common point of breakdown so i thought id ask the people who actually know how it works if im using it wrong: PS4.begin(macAddress) in the setup worked, but seems to conflict? is it worth trying to pin it to a different core? if anyone already has a cam firmware that can give me at least 4 frames a second at any resoloution with a controller connected, id be delighted

pablomarquez76, 2025-02-14:
Hi, I didn’t use the PS4 library with esp32-cam because there are conflicts. The camera library uses so many resources of the ESP32 that is very difficult, at least for me, to find the piece of code that causes the ESP to crash. An alternative that works is to use a web server on the ESP with a web that captures Gamepad Events. Then you can use any device and control your thing using any controller, not only a PS4.  Then, these events (joystick axis movements or buttons pressed) can be converted to actions that can move a robot, and can be transmitted though a websocket server or UDP. I hope this helps

drunkscientist, 2025-02-14:
ok well i feel better at least. do you mean have the controller connected to the viewing device? i really wanted to avoid that for a bunch of reasons, but mostly because its another tier of complication for something that im almost willing to call close enough: if i could get both working not necessarily together but in the same sketch. like if i could turn off the controller and see, then turn the controller on again and move a bit more... whats strange is the esp isnt crashing, and in fact isnt even throwing errors, its saying server ready and controller connected etc, it just wont stream anything while the bluetooth is open. i was able to delay ps4.begin, but when i tried adding PS4.end() to an unused button, it doesnt end the connection. the controller stays on and the camera off

pablomarquez76, 2025-02-14:
Yes, the controller connected to the viewing device. What you describe seem to be a bug in the library, I may have to look into it. If you try the code for the PS4 library alone, without library of the camera, does the code work?

drunkscientist, 2025-02-16:
so first off, thank you for making this library in the first place. second, thank you for taking the time to look into this. i created a new blank sketch or whatever the platformio equivalent is, and  
`if(PS4.event.button_down.share){`
`    PS4.end();`
`    Serial.println("share down, ending PS4 connection");`
`    vTaskDelay(10);`
 ` }`
does not terminate the connection, but it does send the serial line saying it should.