pablomarquez76/PS4_Controller_Host #11: Library breaks with ESP32 release 3.2.0 ?

mnemocron, 2025-04-19:
I wanted to let you know that the latest ESP32 release 3.2.0 breaks the library as outlined in another issue: https://github.com/espressif/arduino-esp32/issues/11262

I don't know who would be responsible to fix it (if at all).

pablomarquez76, 2025-04-22:
Thanks for letting me know. I have to look at the issue very carefully, my guess is that in the new release they have deprecated or modify some of the functions related to bluetooth. Commenting lines 59 and 244 of ps4.c the library can be compiled but you will miss some features related to disconnecting and set the mac address of your ESP32.

pablomarquez76, 2025-05-02:
I made changes needed to fix the issue: https://github.com/espressif/arduino-esp32/issues/11262 
in release 1.0.9 . It should be available soon.

pablomarquez76, 2025-05-03:
> I wanted to let you know that the latest ESP32 release 3.2.0 breaks the library as outlined in another issue: [espressif/arduino-esp32#11262](https://github.com/espressif/arduino-esp32/issues/11262)
> 
> I don't know who would be responsible to fix it (if at all).

The library is already working version 1.1.0 (1.0.9 had some issues with the ESP32 library release so I deleted it)